disk91/WioLoRaWANFieldTester #10: Issue compiling

wbefield, 2021-08-19:
I have edited keys.h with my DevEUI, AppEUI, and AppKey. When I compiled the code I'm getting this  error message.  

`'exit status 1
'processLoRaConfig' was not declared in this scope'`

It compiled perfectly and Wio is working properly once I Commented out the following lines of code. 

`// Make sure the LoraWan configuration has been made if not wait for the configuration
  boolean zero = true;
  for ( int i = 0 ; i < 8 ; i++ ) {
    if ( loraConf.deveui[i] != 0 ) zero = false;
  }
  if ( zero ) {
    configPending();
    while (true) {
      processLoRaConfig();
    }
  }`

disk91, 2021-08-20:
It should be fixed in commit 3eb3bf1522bf4a2e9b922755bf934a8b279e32f9 
Please git pull and compile w/o your modifications
Was compiling on my Arduino platform, strange

wbefield, 2021-08-20:
It is compiling now.  That worked.

wbefield, 2021-08-20:
Let me know if you want me to open a new ticket for these two additional issues.
I'm seeing "Lost" quite a bit on the 1st line of info on the WIO Screen.  Is this due to the late packet issue?
 
Also seeing in Helium Console Debug.  I highlighted in **Bold** below.

Downlink: Queued
Event Information
a few seconds ago
{
  "id": "2a2ba17a-b8a7-4df4-84c8-6c1702e17eb2",
  "router_uuid": "02eaa709-342c-4fb6-ae30-026a8e4d950d",
  "category": "downlink",
  "sub_category": "downlink_queued",
  "description": "Downlink queued in last place",
  "fcnt_down": 6,
  ****"payload": "Unsupported unicode escape sequence in payload",****
  "payload_size": 6,
  "port": 2,
  "reported_at": "1629490521538"

closed 2021-09-02 by wbefield