disk91/WioLoRaWANFieldTester #13: AS923 Frequency

ubiru, 2021-09-18:
Hi, just found out about this great solution for testing the lora network.
Can the code be used for AS923 frequency?
I'm currently ordering the Wio Terminal with the lora gps module, just waiting for the shipment.
But i can't find the option for 923 freq. Or can i use the US915 code for AS923? And if that's possible, what kind of changes needed to be edited?

disk91, 2021-09-19:
It is on progress, I'll add it with the Seeed E5 version. I'm currently on it.

disk91, 2021-09-28:
The AS923 has been released for Wio Chassis LoRa E5

closed 2021-09-28 by disk91

ubiru, 2021-09-29:
thanks for this update. Is the link to to the As923 installation broken?

ubiru, 2021-10-06:
what is the max tx power set in your code on the wio terminal for US915? I seems to recall that it was 22 dBm wit SF12 at max. But now when i just try to test the US915, the max tx power become 20dBm with SF10 at max.