disk91/WioLoRaWANFieldTester #15: AS923 max tx power in WioLoRaWANFieldTester

ubiru, 2021-09-29:
The max tx power set up in the wio terminal for AS923 is 16dBm. This shouldn't be limited at 16dBm as some countries in AS923 region may have different regulated EIRP. 
For example, there is a 26dBm regulated EIRP in some South East Asia countries. So if WIO max tx is set as 16dBm, even an 8dbi antenna wouldn't be able to achieve the max EIRP. Even 10 dBi antenna wouldn't too with the reduction in cable loss.
Can the max tx power for WIO for this purpose (yours and seeed version) be revised to 22 dBm for AS923? Since some people may use the stock antenna that come with the LORA/GPS module from Seeed for the WIO terminal (i believe it is only 1.2 dBi. never got the clear answer from Seeedstudio).

ubiru, 2021-10-09:
Hi 0hotpotman0,

You can find the EIRP value for some of the AS923 region countries here in Helium github, which you can find the link of some asian country regulation for LORA AS923 EIRP value.

https://github.com/helium/miner/issues/1105

Already has Singapore, Indonesia, Malaysia, and Thailand. Which you can see that the EIRP for these countries is 400-500 mW (26-27 dBm).

disk91, 2021-11-06:
The max power has been updated to 22dBm for AS923 with a default set to 16dBm. This will be improved later as much as possible