disk91/WioLoRaWANFieldTester #21: can't connect to AS923_2 hotspot

ubiru, 2021-10-14:
I just noticed that the helium hotspots being deployed with AS923_2 is somehow using AS923_1 freq. I'm currently asking about it in helium/poc-discussion discord. 

It's possible there's a bug in AS923 implementation, but don't know comes from helium code or from the manufacturers. Cause both Linxdot & Sensecaps can witness each other here, but under AS923_1 freq channel (which isn't supposed to be). They should be using AS923_2 freq instead.

Can you confirm that the AS923 freq channel (for region 1,2,34) inside the wiolorawanfieldtester code is using this frequency listed here? https://github.com/helium/miner/blob/0bb8841fbeb1051820691d1f39d4e596fe9d5dfd/test/miner_poc_v11_vars.hrl#L58

ubiru, 2021-11-01:
Can you tell me how to change the code and where, specifically for the AS923_2 frequency? I'm gonna test it using this freq: 'AS923_2' => [921.4, 921.6, 921.8, 922.0, 922.2, 922.4, 922.6, 922.8],

Source from here: https://github.com/helium/miner/blob/master/config/sys.config

0hotpotman0, 2021-11-05:
> Can you tell me how to change the code and where, specifically for the AS923_2 frequency? I'm gonna test it using this freq: 'AS923_2' => [921.4, 921.6, 921.8, 922.0, 922.2, 922.4, 922.6, 922.8],
> 
> Source from here: https://github.com/helium/miner/blob/master/config/sys.config

It has been updated for support AS923_2   you need pull your local file then try again.

ubiru, 2021-11-05:
> It has been updated for support AS923_2 you need pull your local file then try again.

i've tried the new files. After finished uploading it using arduino ide, the wioterminal boot very slow, like around 5-8 times the booting time compare to before. The button selection becomes delayed/not too responsive. Can't select to join the network. Even moving the screen display left and right is also lagging. This is similar to when i edited the LoRaComE5.cpp file with adding the AS923_2 freq before this new update.

I noticed that the connection selection is always stucked. So the 'disc' value won't change to 'join' when selected. And after pusing the join (although the indicator always keep being 'disc' shown on the screen, the lagging of all the button is getting more delayed/lagging).

FYI, i tried with seeed board manager 1.8.1 and 1.8.2. But i have to replace the FlashStorage lib from SAMD with the FlashStorage lib from https://github.com/cmaglie/FlashStorage otherwise the compiler will always resulted in error.

0hotpotman0, 2021-11-05:
> i've tried the new files. After finished uploading it using arduino ide, the wioterminal boot very slow, like around 5-8 times the booting time compare to before. The button selection becomes [...]

You just need to draw the uf2 file put the WT driver,  the WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2 file on binaries

ubiru, 2021-11-05:
> You just need to draw the uf2 file put the WT driver, the WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2 file on binaries

There isn't uf2 file for the 1.1 release yet here. But anyhow, disk91 has helped me sorted out the uploading and now still running some tests.

but the FlashStorage lib from seed SAMD 1.8.2 and 1.8.1 (and probably all of the releases) have problems with the compiler, so need to change the FlashStorage lib as i wrote down earlier.

disk91, 2021-11-06:
> i've tried the new files. After finished uploading it using arduino ide, the wioterminal boot very slow, like around 5-8 times the booting time compare to before. The button selection becomes [...]

This is related to the compilation : you need to select SLAVE mode for the WioTerminal in the Tool menu. When the startup is really long it is because the LoRa E5 is not reachable due to the master mode. So you have all the timeout. I'll add later a message on the WIO screen to indicate this kind of error.

0hotpotman0, 2021-11-11:
@ubiru     Is that work on AS923?

disk91, 2021-12-22:
According to the feedback I've got, it sounds OK

closed 2021-12-22 by disk91

ubiru, 2021-12-23:
@0hotpotman0 yes, it is now working.
@disk91 yes, it's working and also the 22dbm max tx power also can be manually set. Thank you very much for this.