disk91/WioLoRaWANFieldTester #25: Retry (and one other question)

AkosLukacs, 2021-12-22:
Hi, 

This is more like a question, not sure if github issue is the best way to, maybe on Discord...

So, retry in case of a normal sensor is totally sensible, but in case of a mapper - since it's moving - the retry can be from a pretty different location. Especially at SF12 it's over two minutes. So I think retry is actually harmful. Of course, one can set it to 0 :)

But removing retry would free up a button: One idea is single click - turn off the screen (to save battery), OK haven't yet measured, but could help if running from a smaller battery. Long press - lock the controls (for example the mapper is in a backpack or bag).

BTW, do you have a HNT address for donations / "thank you"? :)

disk91, 2021-12-22:
Yes, you are right, when moving the retry can generate wrong information into the mapper. I recommend to set the retry to 0 with doing mapping coverage. the other reason is that your have a large chance to not being ack event if received so you lost opportunities to send new point due to duty cycle.

disk91, 2021-12-22:
> do you have a HNT address for donations / "thank you"? :)
better asking this on discord ;)

closed 2022-10-09 by disk91