disk91/WioLoRaWANFieldTester #26: Data output format

nhatfield, 2022-01-11:
Hello,

Im interested in knowing what the default output of the metrics are? IE: when it sends the data payload off to a console like helium console, is it sending the payload in JSON? I want to be able to use my own backend server if possible

Thank you!

disk91, 2022-01-11:
Look at the developper documentation

nhatfield, 2022-01-12:
@disk91 I started there but was still confused on what the output is. I see how you're building the output, but i dont know what that actually returns in terms of formatting. For example, can I essentially point my field tester at my local api and ingest a JSON payload? Or is it just a blob of text that comes through? I want to be able to use my own tools instead of having to use console, or datacake, etc

Thanks

nhatfield, 2022-01-13:
I think i got it, need to send to console first, then decode, then it sends in json format it seems... testing this now

nhatfield, 2022-01-13:
I made it much further now... for some reason though, im not seeing all of the data... any way to troubleshoot this? My JSON payloads are coming through like this:

```
  {
    "id": "0613158e-14c1-431d-bc02-4a14ce711f41",
    "name": "Test",
    "app_eui": "6081F950CC46BAC1",
    "dev_eui": "6081F9E2C5D12D0D",
    "devaddr": "9E030048",
    "downlink_url": "<url>",
    "fcnt": "1170",
    "port": "1",
    "payload": "l0enUzNOBDQJCg==",
    "reported_at": "1642035821645"
  }
  ```

Im using this decode:

```
function Decoder(bytes, port) { 
  var decoded = {};
  
  var lonSign = (bytes[0]>>7) & 0x01 ? -1 : 1;
  var latSign = (bytes[0]>>6) & 0x01 ? -1 : 1;
  
  var encLat = ((bytes[0] & 0x3f)<<17)+
               (bytes[1]<<9)+
               (bytes[2]<<1)+
               (bytes[3]>>7);

  var encLon = ((bytes[3] & 0x7f)<<16)+
               (bytes[4]<<8)+
               bytes[5];
  
  var hdop = bytes[8]/10;
  var sats = bytes[9];
  
  const maxHdop = 2;
  const minSats = 5;
  
  if ((hdop < maxHdop) && (sats >= minSats)) {
    // Send only acceptable quality of position to mappers
    decoded.latitude = latSign * (encLat * 108 + 53) / 10000000;
    decoded.longitude = lonSign * (encLon * 215 + 107) / 10000000;  
    decoded.altitude = ((bytes[6]<<8)+bytes[7])-1000;
    decoded.accuracy = (hdop*5+5)/10
    decoded.hdop = hdop;
    decoded.sats = sats;
  } else {
    decoded.error = "Need more GPS precision (hdop must be <"+maxHdop+
      " & sats must be >= "+minSats+") current hdop: "+hdop+" & sats:"+sats;
  }

  return decoded;
}
```

nhatfield, 2022-01-13:
Figured it, the issue was with the JSON template in console! Thanks @disk91

closed 2022-01-13 by nhatfield