disk91/WioLoRaWANFieldTester #32: LoRa board is missing

ralphjy, 2022-04-14:
Hi,
I just got a Seeed Wio Terminal LoRa-E5 and GNSS Chassis from Digikey.
I flashed it using the V1.6 WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2 firmware.
I configured the frequency (US915) and the DevEui, AppEui, and AppKey.

I get LoRaWan configuration OK, but when the program restarts I get a "LoRa board is missing".
Seeed support doesn't seem to know how to debug this problem.
Can you help me?

Thanks,
Ralph

disk91, 2022-04-15:
I assume you compiled with Wio as MASTER you need to setup as SLAVE in the Tool menu.
Read carefully the DEVELOPER.MD file.

ralphjy, 2022-04-15:
Actually, I had assumed that if I used the precompiled V1.6 WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2 file that it should work.

I'll try downloading the source and make the changes in the DEVELOPER.MD file and compiling.

disk91, 2022-04-16:
Humm… ok misreading from me. Not need to compile. You error was looking a common mistake when people build the imagine on their own. 
Please try the version 1.5 to check if that one is working. I will double check the version 1.6.

ralphjy, 2022-04-16:
I tried version 1.5 with the same result.

I noticed that the error happens in the loraSetup function after the while(!SERIALE5);
Does that imply that the loraE5 was able to connect?  I want to determine where the problem is (if it's just a bad connection).  Maybe I just have a bad module.

disk91, 2022-04-16:
I did check the firmware, no problem on my devices.
Can you compile with following logs activated (in config.h)
#define DEBUG
#define DEBUGLORA
And send me the log you get on the console (please make sure you hide at least the appkey that will be displayed in the log)

ralphjy, 2022-04-16:
I compiled version 1.6 with changes per the DEVELOPER.md.  And activated the DEBUG and DEBUGLORA logs.

Here is the output
`08:47:33.458 -> LoRa timeout
08:47:33.458 -> AT
08:47:35.449 -> LoRa timeout
08:47:35.449 -> AT
08:47:37.447 -> LoRa timeout
08:47:37.447 -> AT
08:47:39.448 -> LoRa timeout
08:48:03.213 -> ZONE: US915
08:48:03.213 -> OK
08:48:15.508 -> DEVEUI:70B3D57ED004EBE9
08:48:15.508 -> OK
08:48:28.183 -> APPEUI:0000000000000000
08:48:28.183 -> OK
08:48:40.845 -> APPKEY:XXXXXXXXXXXXXXXXXXXXX
08:48:40.845 -> OK
08:48:40.878 -> LoRaWan configuration OK`

Then it stops at the Splash screen with the Banner "LoRa board is missing.

disk91, 2022-04-16:
So your LoRa board is not responding. 
i assume there is a problem with the hardware. It basically never respond.

ralphjy, 2022-04-16:
Okay thanks.  You can close this.  I don't think I want to take it apart to mess with it.  That might prevent me from returning it.  I am going to ask for an RMA.

I did try a retstart and it appears that It can see the GPS.

08:39:40.372 -> AT
08:39:42.348 -> LoRa timeout
08:39:42.348 -> AT
08:39:44.375 -> LoRa timeout
08:39:45.834 -> Restart
08:39:45.954 -> Done
08:39:45.954 -> L86 Found
08:39:48.922 -> AT
08:39:50.894 -> LoRa timeout
08:39:50.894 -> AT
08:39:52.910 -> LoRa timeout

Thank you for all your great work.  It's too bad my hardware isn't working.

Ralph

ralphjy, 2022-04-16:
It appears that the problem is not the LoRa board, but Serial1 on my Wio Terminal.  I was able to get another Wio Terminal to try and things are now working.  I need to see if I can determine what the problem with Serial1 is.  I don't normally use the RPi connector or Serial1, so this may have always been broken...

Thanks again for your help.

disk91, 2022-04-19:
good to know that you found the hardware problem. 
Don't know what can be you serial port Issue, should be hardware. The good news is that it is easier to find a Wio than a LoRaE5 chassis

closed 2022-04-19 by disk91