disk91/WioLoRaWANFieldTester #34: Seeed bundle - change config

leogaggl, 2022-04-21:
Is it possible to change the Seeed bundle to your own Helium DEVEUI and keys?

The serial commands do not appear to work?

Apologies if this is a FAQ - but could not find info.

disk91, 2022-04-21:
Important point : if you follow the procedure, you will lost the ability to get back and get the benefit of the bundled communication. Then, you will have to reconfigure the whole helium console to get benefit of screen display, mapper & cargo.

If you still want to connect to helium console with your own account:
1) upload the [Clear_LoRaE5_eprom.uf2](https://github.com/disk91/WioLoRaWANFieldTester/blob/master/binaries/Clear_LoRaE5_eprom.uf2) on your Wio
2) start the Wio terminal for about 30 seconds
3) upload the [WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2](https://github.com/disk91/WioLoRaWANFieldTester/blob/master/binaries/WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2)
4) you can setup your device and helium console as described in the [SETUP.md](https://github.com/disk91/WioLoRaWANFieldTester/blob/master/doc/SETUP.md)

If you have any difficulty to understand what is described in this message, the best is to not process this steps. You won't have any support to execute it.

leogaggl, 2022-04-21:
Thank you for the prompt answer, Paul!

Much appreciated. Exactly what I am after. I do want to ingest the data into another DB as well. I am comfortable with configuring things myself. I was actually expecting the unit to need setup rather than some black box.

Keep up the great work!

disk91, 2022-04-21:
Great. 
All the comments were not for you: I made the response to be shared as it is a common question ;)