disk91/WioLoRaWANFieldTester #35: FlashStorage.cpp - NVMCTRL_CTRLB_Type::<unnamed>' has no member named 'MANW'

Kawondo, 2022-04-27:
Hi, 

I've followed the instructions exactly, but FlashStorage.cpp doesn't compile on SAMD51 platforms (Wio Terminal) because the defilnition NVMCTRL->CTRLB.bit.MANW is missing. Tested with Wio development environment 1.8.2, 1.8.1, 1.7.9 on Windows and OSX (clean install). Did I miss something?

KR
...
/Users/devel/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/libraries/FlashStorage/src/FlashStorage.cpp: In member function 'void FlashClass::write(const volatile void*, const void*, uint32_t)':
/Users/devel/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/libraries/FlashStorage/src/FlashStorage.cpp:55:22: error: 'volatile struct NVMCTRL_CTRLB_Type::<unnamed>' has no member named 'MANW'
   NVMCTRL->CTRLB.bit.MANW = 1;
                      ^~~~

disk91, 2022-05-02:
Make sure you completely followed the DEVELOPER.md documentation. Until now, all people having compiled the project succeed in it.

Kawondo, 2022-05-02:
Hi,
The problem is that the FlaschStorage library (v.0.5.0) supplied with the WIO toolchain is not compatible with a SAMD51 chip. There is a more modern one (v.1.0.0) specifically for the D51 on github, so it works without any problems: https://github.com/cmaglie/FlashStorage (A convenient way to store data into Flash memory on the ATSAMD21 and ATSAMD51 processor family)

...
Multiple libraries were found for "FlashStorage.h"
  Used: /Users/devel/Documents/Arduino/libraries/FlashStorage
  Not used: /Users/devel/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/libraries/FlashStorage

closed 2022-10-09 by disk91