disk91/WioLoRaWANFieldTester #39: Flash update process ??

vincegre, 2022-07-05:
Hi

Just got a Wio Tracker Lora kit but I'm quite confused if it needs to be updated. It states version 1.5 for E5 when it boots on the Wio. I have seen some 1.8 version here. Is it a good idea to update ? Should I also before flashing 1.8 load the fw to clear E5 configuration ? or can I flash straight 1.8 version ?

Thanks

Vincèn

JerryTN1, 2022-07-10:
I have recently purchased the Wio Seeed package where it came with monitor and antenna module.  Arrived with version 1.5.  Saw a video about the update to 1.8 to add discovery feature and fix some issues.  I went ahead and did the update to version 1.8 as saw where it shouldn't cause any issues with the credentials then later with more research saw where that was in version 1.7  After the update for 1.8 the device booted up looking normal and had the additional discovery option after scrolling pass the GPS option.  Problem I am having at the moment is so far has not join any hot spot. GPS is working normal but no reports since it hasn't join the network making me think the credentials may have been lost and will now have to manually setup the device. So at the moment I am hoping someone will respond to your question and a possible fix for me or work around on how to get the device back to joining the network.  Will update if something changes and starts working again or if I figure out what needs to be done.   Before the update I did go out in the car and the mapping feature seems to work great and can be seen on mapper and cargo.  Only thing is not sure who I need to add this device to console so I can get reports on miners it is seeing so I can get a better knowledge of what is it seeing.

disk91, 2022-07-10:
Sorry i missed the issue creation, @vincegre 
Yes you can update and yes this preserve your settings. The version 1.8 has a little bug that do not prevent to join but that prevent to map. If you have it : join and then run a disco mode after this the device will work correctly. I recommend to downgrade to 1.7 for now. 
@JerryTN1 what 2 files are you refering for the upgrade ? There is only one that can be use. Let le know to see if i can clarify the procedure. 

About the discovery mode : it shows the hotspot that relayed the traffic for 10 frames with a limit of 4 hotspot per frame. If sole hotspot are responding late or are not lucky on random selection they won’t be displayed. This can be the reason why you don’t see all between. There is two ways to show more:
- run a second disco mode after the previous one (data are cumulated within an hour)
- Attach Wio terminal to your own console account and setup unlimited buy to get all the duplicates. (You lost the advantage of ready to go and paid DCs)

closed 2022-09-01 by disk91