disk91/WioLoRaWANFieldTester #42: device field tester with Kerlink Wansey

t0mZ4, 2022-07-26:
Hi there

we'd like to use the field tester with Kerlink WMC and TTI

We have the SEEED bundle.  

Can we get the appKey and just add them to our network servers? Or do we need to flash different firmware?

Also, where do we get the appKey per-defined in the seeed bundle? 

thanks alot!

disk91, 2022-07-26:
The version 1.9 of the firmware is compatible with TTN. 
Then you need to configure TTN / TTI like it is described in the github project page. 
If you have a seeed bundle, preconfigured with Helium credentials, you need to follow the github issue « change config to manage my own console account .. » you will lost the helium credentials forever. 
Paul

_______________________________
Paul Pinault
http://www.disk91.com
http://www.twitter.com/disk_91

t0mZ4, 2022-08-03:
thanks alot! so I am trying to flash the firm to 1.9 [All Zone LoRa E5 Chassis (Seeed)](https://github.com/disk91/WioLoRaWANFieldTester/blob/master/binaries/WioLoRaWANFieldTester_LoRaE5_ALLZONE.uf2)

I see the arduino drive mounting, i drag the file, nothing happens. I am using linux.

anything else should do or try?

disk91, 2022-08-03:
I never try it with Linux. 
You should try to flush the drive with sync command after your copy. 

_______________________________
Paul Pinault
http://www.disk91.com
http://www.twitter.com/disk_91

t0mZ4, 2022-08-03:
managed to flash it fine from windows laptop, thanks!

closed 2022-08-03 by t0mZ4