disk91/WioLoRaWANFieldTester #45: Wio Terminal Feild tester

DaGemm, 2022-07-28:
Received field tester from Seed Studio.  It worked fine for about 1 month and the screen stopped working. Both lights are on when connected to a usb c power cord.  Here was the email that I received from Seed Studio. 

I see that the blue light of your device is flashing quickly. You can quickly dial the power button on the left twice to make the device enter the Bootloader mode, and then dial it again to exit the bootloader mode, and the device should be restored to normal.

If the firmware is still not displayed, please try to refresh the firmware according to this wiki. Remember to remove the Lorawan expansion board when refreshing the firmware and refresh wiO Terminal only.

When I use Audruino and try to select port with terminal I get this msg.

Programmer not responding.  Make sure that the firmwareupdater sketch is loaded on the board.  

Any help would be appreciated.

disk91, 2022-07-28:
Did you try to enter dfu mode (double switch reset) the drag & drop the uf2 binary from github to the Arduino drive appeared when in dfu mode ?

disk91, 2022-08-15:
no more elements

closed 2022-08-15 by disk91