disk91/WioLoRaWANFieldTester #47: Does Discovery mode support TTN?

HAO156, 2022-08-09:
**Does Discovery mode support TTN? Or does it only support Helium?** 
  When I use Discovery mode in the TTN environment.The link in the QR code cannot be accessed from a mobile phone or computer.
[image]

  Any help would be appreciated.

disk91, 2022-08-09:
Discovery mode is not yet implemented with TTN. 
It will be added soon. 

_______________________________
Paul Pinault
http://www.disk91.com
http://www.twitter.com/disk_91

HAO156, 2022-08-10:
Thank you for the reply. Looking forward to supporting the discovery mode of the TTN version.

closed 2022-08-10 by HAO156