disk91/WioLoRaWANFieldTester #49: Error compiling to Seeduino Wio Terminal

aruznieto, 2022-08-16:
Hi, I am trying to compile the software in Arduino. When I set the Role to Slave, I generate the binaries but Arduino´s Software return me this error: "Error compiling for Seeeduino Wio Terminal board." This is my configuration on Tools

[image]

Thanks!

closed 2022-08-16 by aruznieto

disk91, 2022-08-16:
What has been you solution ?

aruznieto, 2022-08-16:
Adding 

`&& defined FALSE`

on "Libraries/Arduino15/packages/seeeduino/hardware/samd/1.8.2/library/Wire/Wire.cpp" next to 

`#if WIRE_INTERFACES_COUNT > 1`

MaffooClock, 2025-06-24:
I just ran into this today (using PlatformIO).

The full error is about SERCOM functions being defined twice:
```text
.pio/build/seeed_wio_terminal/lib484/libWire.a(Wire.cpp.o): In function `SERCOM4_0_Handler':
Wire.cpp:(.text.SERCOM4_0_Handler+0x0): multiple definition of `SERCOM4_0_Handler'
.pio/build/seeed_wio_terminal/libFrameworkArduinoVariant.a(variant.cpp.o):variant.cpp:(.text.SERCOM4_0_Handler+0x0): first defined here
.pio/build/seeed_wio_terminal/lib484/libWire.a(Wire.cpp.o): In function `SERCOM4_1_Handler':
Wire.cpp:(.text.SERCOM4_1_Handler+0x0): multiple definition of `SERCOM4_1_Handler'
.pio/build/seeed_wio_terminal/libFrameworkArduinoVariant.a(variant.cpp.o):variant.cpp:(.text.SERCOM4_1_Handler+0x0): first defined here
.pio/build/seeed_wio_terminal/lib484/libWire.a(Wire.cpp.o): In function `SERCOM4_2_Handler':
Wire.cpp:(.text.SERCOM4_2_Handler+0x0): multiple definition of `SERCOM4_2_Handler'
.pio/build/seeed_wio_terminal/libFrameworkArduinoVariant.a(variant.cpp.o):variant.cpp:(.text.SERCOM4_2_Handler+0x0): first defined here
.pio/build/seeed_wio_terminal/lib484/libWire.a(Wire.cpp.o): In function `SERCOM4_3_Handler':
Wire.cpp:(.text.SERCOM4_3_Handler+0x0): multiple definition of `SERCOM4_3_Handler'
.pio/build/seeed_wio_terminal/libFrameworkArduinoVariant.a(variant.cpp.o):variant.cpp:(.text.SERCOM4_3_Handler+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/seeed_wio_terminal/firmware.elf] Error 1
```

[Andrés's solution](#issuecomment-1217153673) is one of several ways to work around this.  Personally, I opted to just comment out those four definitions in [Wire.cpp](https://github.com/Seeed-Studio/ArduinoCore-samd/blob/d900ef7e959df763a4f56338183964739790f035/libraries/Wire/Wire.cpp#L307-L310):
```cpp
// Line 306 - 311
  #if defined(__SAMD51__)
    // void WIRE1_IT_HANDLER_0(void) { Wire1.onService(); }
    // void WIRE1_IT_HANDLER_1(void) { Wire1.onService(); }
    // void WIRE1_IT_HANDLER_2(void) { Wire1.onService(); }
    // void WIRE1_IT_HANDLER_3(void) { Wire1.onService(); }
  #endif // __SAMD51__
```

I spent a few moments looking through the framework source to try to figure out why this was happening, but I gave up.  Also checked the [Seeed-Studio/ArduinoCore-samd](https://github.com/Seeed-Studio/ArduinoCore-samd) repo to see if anyone filed an issue for this, but there is none.  And I didn't do enough of my own homework to feel confident enough to open an issue.