disk91/WioLoRaWANFieldTester #52: In Bootloader Mode

Meyblaubaer, 2022-08-25:
Hello,

i got the Seeed Package. After Updating to newest Firmware, The Device keeps booting in bootloader mode. I tried reinstalling firmware without success.

disk91, 2022-08-25:
That is strange and unusual. What firmware file did you flashed ?

_______________________________
Paul Pinault
http://www.disk91.com
http://www.twitter.com/disk_91

Meyblaubaer, 2022-08-26:
I was able to resolve the failure. It was because an faulty USB-C Cable, something went wrong. After uploading the Arduino "Blink Example" i was able to reupload with v1.3 and after this to the actual firmware. So i think it was a fault on my side, and this issue can be closed.

closed 2022-08-26 by Meyblaubaer