disk91/WioLoRaWANFieldTester #53: Flashed 1.9 to Seeed Wio, stuck at startup screen

circuitmike, 2022-09-01:
I got the Seeed Wio bundle a while back and thought it was time for a firmware update. I flashed 1.9 (LoRaE5) to the Wio and now it simply hangs at the startup screen ("Wio LoRaWan Field Tester / Version v1.9 (LoRaE5) / WIO_FT_XXXXXXXXXX"). On the LoRa section there's a steady blue light and blinking red light. Any idea what might be going on?

disk91, 2022-09-01:
Hello, 
That looks strange ...
I propose you flash it again ...

If you stil have the same issue please DM me on discord we will investigate this.

circuitmike, 2022-09-04:
Flashing again was the first thing I tried, and it did exactly the same thing. I'll try to find you on Discord one of these days when I've got some time.

closed 2022-09-25 by disk91

circuitmike, 2022-10-21:
I'm leaving this closed because I finally figured out I had a hardware problem. The 5-way switch on the front of my Wio Terminal has failed and at least one of the 5 switches is constantly bouncing. This causes the device to go into and immediately out of configuration mode at startup, causing a boot loop. Once in a while if I hold the button just right I can get it to start up, but that's it. I got a new Wio Terminal but unfortunately I don't see a way to recover the app key provided by Seeed.

disk91, 2022-10-21:
I can help you on this. DM me on discord