disk91/WioLoRaWANFieldTester #57: Decoding data from TTN to DataCake

richwriroo, 2022-09-17:
I am having trouble decoding data on DataCake 

I have it set up on TTN which works great and have set up an integration with DataCake - I tried your decoder from your instructions with DataCake but don't get any correct values for any decoded values.  My integration webhook URL from TTN is https://api.datacake.co/integrations/lorawan/tti

Here is the decoder I am using:

`function Decoder(bytes, port) { 
  var decoded = {};
  // avoid sending Downlink ACK to integration (Cargo)
  if (port === 1) {
    var lonSign = (bytes[0]>>7) & 0x01 ? -1 : 1;
    var latSign = (bytes[0]>>6) & 0x01 ? -1 : 1;
    
    var encLat = ((bytes[0] & 0x3f)<<17)+
                 (bytes[1]<<9)+
                 (bytes[2]<<1)+
                 (bytes[3]>>7);
  
    var encLon = ((bytes[3] & 0x7f)<<16)+
                 (bytes[4]<<8)+
                 bytes[5];
    
    var hdop = bytes[8]/10;
    var sats = bytes[9];
    
    var maxHdop = 2;
    var minSats = 5;
    
   // if ((hdop < maxHdop) && (sats >= minSats)) {
      // Send only acceptable quality of position to mappers
      decoded.latitude = latSign * (encLat * 108 + 53) / 10000000;
      decoded.longitude = lonSign * (encLon * 215 + 107) / 10000000;  
      decoded.altitude = ((bytes[6]<<8)+bytes[7])-1000;
      decoded.accuracy = (hdop*5+5)/10
      decoded.hdop = hdop;
      decoded.sats = sats;
   // } else {
   //   decoded.error = "Need more GPS precision (hdop must be <"+maxHdop+
        " & sats must be >= "+minSats+") current hdop: "+hdop+" & sats:"+sats;
   // }
    return decoded;
  }
    return null;
}`

I get something that looks like this when debugging in DataCake:

`{
    "accuracy": 0.5,
    "altitude": -1000,
    "hdop": 0,
    "latitude": 5.3e-06,
    "longitude": 1.07e-05,
    "sats": 0
}`

disk91, 2022-09-17:
Please give me your original payload (before decodind)
I think this is a payload from a frame w/o gps position (basically full of 0)

richwriroo, 2022-09-18:
Hi Paul thanks for your speedy reply - here is a screen grab from TTN - I think you may be right about the payload being zero's - is my config wrong.
[image]

disk91, 2022-09-25:
You get 00.. when there is no GPS fix
You decoder should bypass this situation.