disk91/WioLoRaWANFieldTester #59: How to use with TTN ?

vincegre, 2022-09-25:
Hi

I'm trying to follow your documentation on how to configure it with TTN but It's not enought detailesd:

`The Application EUI is by default set to 00..00, you need to change this to anything else to be able to configure your WioLoRaWanFieldTester.`

How do I change that in the Wio ? does it mean I have to recompile your code with my own AppEUI ?

Thanks

Vincèn

aruznieto, 2022-09-25:
Hi,

you can recompile writing your keys on [`keys.h` ](https://github.com/disk91/WioLoRaWANFieldTester/blob/master/keys.h) or restart the device pressing the 5 direction button. After 1 second after rebooting, stop pushing the 5 direction button, you should see the setup screen displayed and you can change the keys.

vincegre, 2022-09-25:
Thanks and i'll push an update on the documentation as it's missing that little detail ;) I just have one little issue: how do I change the AppKey as per default it's XXXXXXX: I tried left/right and also the top 3 buttons but none of them allows to edit it :(

vincegre, 2022-09-25:
Well I went to edit in code in Arduino IDE but I don't succeed to compile your project as downloaded from github :( I added missing libs in Arduino IDE and putted keys in keys.h but when I compile I get that :(

```
Compilation du croquis...
"/usr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/usr/share/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/share/arduino/hardware/arduino/avr/variants/standard" "-I/usr/share/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/vincen/Arduino/libraries/TFT_eSPI" "-I/home/vincen/Arduino/libraries/SparkFun_BQ27441_LiPo_Fuel_Gauge_Arduino_Library/src" "-I/home/vincen/Arduino/libraries/Adafruit_GPS_Library/src" "-I/usr/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial/src" "-I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/src" "-I/home/vincen/Arduino/libraries/FlashStorage/src" "-I/home/vincen/Arduino/libraries/QRCode/src" "/tmp/arduino_build_672054/sketch/LoRaCom.cpp" -o "/tmp/arduino_build_672054/sketch/LoRaCom.cpp.o"
/tmp/arduino_build_672054/sketch/LoRaCom.cpp: In function ‘bool processLoRaConfig()’:
/tmp/arduino_build_672054/sketch/LoRaCom.cpp:454:30: error: ‘class HardwareSerial’ has no member named ‘printf’
                 SERIALCONFIG.printf("%02X",loraConf.deveui[i]);
                              ^
/tmp/arduino_build_672054/sketch/LoRaCom.cpp:478:30: error: ‘class HardwareSerial’ has no member named ‘printf’
                 SERIALCONFIG.printf("%02X",loraConf.appeui[i]);
                              ^
/tmp/arduino_build_672054/sketch/LoRaCom.cpp:502:30: error: ‘class HardwareSerial’ has no member named ‘printf’
                 SERIALCONFIG.printf("%02X",loraConf.appkey[i]);
                              ^
/tmp/arduino_build_672054/sketch/LoRaCom.cpp:610:22: error: ‘NVIC_SystemReset’ was not declared in this scope
     NVIC_SystemReset();
                      ^
Utilisation de la bibliothèque SPI version 1.0 dans le dossier: /usr/share/arduino/hardware/arduino/avr/libraries/SPI 
Utilisation de la bibliothèque TFT_eSPI version 2.4.72 dans le dossier: /home/vincen/Arduino/libraries/TFT_eSPI 
Utilisation de la bibliothèque SparkFun_BQ27441_LiPo_Fuel_Gauge_Arduino_Library version 1.1.0 dans le dossier: /home/vincen/Arduino/libraries/SparkFun_BQ27441_LiPo_Fuel_Gauge_Arduino_Library 
Utilisation de la bibliothèque Adafruit_GPS_Library version 1.7.0 dans le dossier: /home/vincen/Arduino/libraries/Adafruit_GPS_Library 
Utilisation de la bibliothèque SoftwareSerial version 1.0 dans le dossier: /usr/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial 
Utilisation de la bibliothèque Wire version 1.0 dans le dossier: /usr/share/arduino/hardware/arduino/avr/libraries/Wire 
Utilisation de la bibliothèque FlashStorage version 1.0.0 dans le dossier: /home/vincen/Arduino/libraries/FlashStorage 
Utilisation de la bibliothèque QRCode version 0.0.1 dans le dossier: /home/vincen/Arduino/libraries/QRCode 
exit status 1
Erreur de compilation pour la carte Arduino Uno

```

disk91, 2022-09-25:
Look like you did not followed the DEVELOPEMENT.md steps.
You don't really need to change the key.h file and recompile, the easiest way is to use the configuration over the serial port documented in SETUP.md

vincegre, 2022-09-25:
> Look like you did not followed the DEVELOPEMENT.md steps.
> You don't really need to change the key.h file and recompile, the easiest way is to use the configuration over the serial port documented in SETUP.md

Thanks a lot, I had missed these docs, sorry for that :( Unhappy I'll have try again an other day, looks like I'm very unlucky on technology is agains me today :( Compilation still doesn't work... The setup in the device itself at boot is not working for Appkey (the 2 other fields have no problem to be edited)... setup using serial through USB neither :(

disk91, 2022-09-25:
If you have *** for the appkey you first need to remove the bundle protection. For this take a look at the issues you will find one indicating how to do it. _______________________________Paul Pinaulthttp://www.disk91.comhttp://www.twitter.com/disk_91

vincegre, 2022-09-26:
> If you have *** for the appkey you first need to remove the bundle protection. For this take a look at the issues you will find one indicating how to do it.

Thanks, problem at least solved so could properly setup keys in device and now it sends payload at TTN ;)
It just doesn't map anything on ttnmapper and after a while displayed a QRCode on screen but none of my mobile devices recognise the code and accepts to scan it :(
I think I'm going to try to find out if someone published some codes for Wio Terminal to use it properly as a mapper as I have with Lilygo ESP/Lora devices !

Thanks again for all your support ;)

disk91, 2022-09-26:
The data are automatically reported to TTN Mapper if you correctly setup the integration to the dev.disk91.com backend.
To get the data displayed on TTN Mapper you need to have a GPS coverage and a suficient Hdop quality (precision) basically the GPS dot status must be green.

What you describe with the QR code is the Discovery Mode, is is use to get the global coverage around but sending multpli message. This is not a mapping feature. The discovery mode is not implemented on TTN yet. For mapping, just use the standard mode.

I don't know why the QR code not working in your case, It works well with many of the smartphone default photo app. You can eventually try with a dedicated QRCode reader app. Nothing specific with that QR.

vincegre, 2022-10-01:
> The data are automatically reported to TTN Mapper if you correctly setup the integration to the dev.disk91.com backend. To get the data displayed on TTN Mapper you need to have a GPS coverage and a suficient Hdop quality (precision) basically the GPS dot status must be green.

I double-checked the whole setup but I still get an error when outside with green GPS in TTN console when it sends payloads from Wio :( Webhook should be setup that way right ?
[image]
Next time I go out with Wio I'll capture exact error message in console to find out where problem is !

> 
> What you describe with the QR code is the Discovery Mode, is is use to get the global coverage around but sending multpli message. This is not a mapping feature. The discovery mode is not implemented on TTN yet. For mapping, just use the standard mode.

Thanks for clarification ;)