disk91/WioLoRaWANFieldTester #61: Feature Request: add a setting or default that makes the screen turn off after a certain timeout period

DeflateAwning, 2022-09-28:
It would be great if the screen naturally turned off after 5-10 minutes of no button presses, then turned back on when you press a button. 

This setting could be enable by either:
* a config setting in the code (set at compile-time)
* a setting accessible on the device via the buttons, perhaps on a new "Settings" page

The screen uses energy and is very bright when there is no ambient light, so it would be much nicer if it didn't use power to be so intrusive.

disk91, 2022-09-28:
Hello,I’m planning to have a mode where button are not active and screen will be off for mapping during walk. So it will be in this. I dislike the auto off, as it exists on the rak version, setup page would be a harder work currently. Thank you for sharing this idea. _______________________________Paul Pinaulthttp://www.disk91.comhttp://www.twitter.com/disk_91

DeflateAwning, 2022-09-28:
That sounds great. With that enabled, how do you intend to be able to turn the screen back on? Will that be possible?

Also, it works better when you reply using the online Github interface instead of replying to the email. Your last response got a bit garbled.

disk91, 2022-10-09:
Added in master, will be in the coming release 1.10

closed 2022-10-09 by disk91