disk91/WioLoRaWANFieldTester #62: Configuring The Seed device

BrettOT, 2022-10-03:
You have probably been asked this before but i need some clarification.
1. I need to set the Application key on the device to get it to connect. When i look in the setup it is x'd out.
    I can set it using putty commands direct to the module but, when power is removed this does not save.
2. We are using AU915 and it always defaults to sub band 1. We use sub band 8 for most instances.
    As with the application key i can set the frequency band via direct putty commands to the module and then it works.
    Again though when power is removed, the unit defaults. Is there a way i can write the commands i want into the unit and have them
    stay there?

disk91, 2022-10-03:
Hello 
1 - take a look at the issue #34 
2 - The subband selected is currently 2 (channel 8 to 15), I can add a AU915_8 mode with subband 8 if you give me the corresponding channels. Please open a specific Issue with the request to track it and implement it in the coming release

disk91, 2022-10-11:
I need you feed back on subbands to include this in the version 1.10 
sooner is better, later means next release

BrettOT, 2022-10-11:
It would be great if after selecting a country that has sub bands if you could then select the sub band you wish to use.
For the US and AU915 the sub bands are as follows:
1: 0-7, 64
2: 8-15, 65
3: 16-23, 66
4: 24-31, 67
5: 32-39, 68
6: 40-47, 69
7: 48-55, 70
8: 56-63, 71

The most common we use is AU915 sub band 8 however, can be any due to local interference.
We also use the US frequencies and sub bands as we sell our product all over the world as well as the EU and India frequencies.

The idea is to use this as a cheap portable test unit for field strength testing.

Regards,

Brett O'Toole
Communications Technician
Direct: +61358208846
Mobile: +61429138190
***@***.******@***.***>

Rubicon Water

Shepparton Office
1a Wheeler Street, Shepparton
Victoria 3630, Australia
Tel +61 3 5820 8800
www.rubiconwater.com
***@***.***

[LinkedIn]<https://www.linkedin.com/company/861213/>  [Facebook] <https://www.facebook.com/RubiconWater/>   [Instagram] <https://www.instagram.com/rubiconwater/>   [Twitter] <https://twitter.com/RubiconWater>   [YouTube] <https://www.youtube.com/user/rubiconsystemsau>

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived by Mimecast Ltd, an innovator in Software as a Service (SaaS) for business. Providing a safer and more useful place for your human generated data. Specializing in; Security, archiving and compliance. To find out more visit the Mimecast website.

t0mZ4, 2022-10-19:
We'd like to +1 this and also have the sub selection for AU915 and US915 
we use FSB8 in USA currently.

disk91, 2022-10-27:
This has been added to master
Please give a try before release creation

BrettOT, 2022-10-27:
Where do I get the code from?

Regards,

Brett O'Toole
Communications Technician
Direct: +61358208846
Mobile: +61429138190
***@***.******@***.***>

Rubicon Water

Shepparton Office
1a Wheeler Street, Shepparton
Victoria 3630, Australia
Tel +61 3 5820 8800
www.rubiconwater.com
***@***.***

[LinkedIn]<https://www.linkedin.com/company/861213/>  [Facebook] <https://www.facebook.com/RubiconWater/>   [Instagram] <https://www.instagram.com/rubiconwater/>   [Twitter] <https://twitter.com/RubiconWater>   [YouTube] <https://www.youtube.com/user/rubiconsystemsau>