disk91/WioLoRaWANFieldTester #63: Add Sub-band selection to US915/AU915

t0mZ4, 2022-10-19:
referencing to https://github.com/disk91/WioLoRaWANFieldTester/issues/62

we'd like to request the option to select sub-band for AU915 and US915.

US915 - 

Sub-Band 1 | 902.3 - 903.7 | 0-7
Sub-Band 2 | 903.9 - 905.3 | 8-15
Sub-Band 3 | 905.5 - 906.9 | 16-23
Sub-Band 4 | 907.1 - 908.5 | 24-31
Sub-Band 5 | 908.7 - 910.1 | 32-39
Sub-Band 6 | 910.3 - 911.7 | 40-47
Sub-Band 7 | 911.9 - 913.3 | 48-55
Sub-Band 8 | 913.5 - 914.9 | 56-63

Downlink sub-bands | Frequency range (MHz) | Channels
Downlink sub-band | 903 - 914.2 | 64-71

(reference: https://www.baranidesign.com/faq-articles/2019/4/23/lorawan-usa-frequencies-channels-and-sub-bands-for-iot-devices)

t0mZ4, 2022-10-25:
@disk91 any prediction on when this might be incorporated in a release?

kind regards

disk91, 2022-10-25:
Next release, in progress. I hope mid November

disk91, 2022-10-27:
It has been added to master.
Tester welcome

mcauser, 2022-11-21:
FYI AU915 as of 2022-11-17 has switched from FSB2 (ch8-15) to FSB6 (ch40-47). They're calling it Dual Plan so joins can overlap with AS923 devices.

deeeblack, 2022-11-21:
out of curiosity, how does one select the sub band? GUI or UART?

mcauser, 2022-11-21:
@deeeblack
GUI:
Reboot, hold middle d-pad button until you see text, release button
Presents the config screen
Use top buttons to cycle through zones

UART:
The zone is in the EEPROM at address 0x29
If you modify it, you'll need to recalculate the checksum at 0x03
Easier just using the GUI.

I just found/fixed a bug with AU915 sub band selection: https://github.com/disk91/WioLoRaWANFieldTester/pull/65
Missed 2 vars in the search/replace.

deeeblack, 2022-11-23:
Of note, I can only select +16dbm for power when using the AU915-6 setting. Can we max the power I think +22dbm on this setting?