disk91/WioLoRaWANFieldTester #64: Wio-terminal does not boot with E5 firmware.

theoldinternet, 2022-11-06:
I have wio-terminal with the E5 Lorawan module.  The device does not boot after I drag the E5 .u2f over to the wio-terminal drive. 
The RFM version will boot, but obviously lorawan won't work because it's not the correct module.

disk91, 2022-11-06:
Make sure you have selected the LORA E5 version and reload it. Sometime it crash on the first time, I already seen this.
Let me know if it continue. Eventually check if it works better with version 1.9 in case the new version 1.10 has some problem (I'll double check also)

theoldinternet, 2022-11-06:
I turns out that it will not boot while the lorawan module is attached.  If you attach it before you see the first graphic, it works fine