disk91/WioLoRaWANFieldTester #72: TTN webhook configuration for WioLoRaWANFieldTester

ffries, 2023-04-09:
Dear community,

Thank you for working on WioLoRaWANFieldTester.

I purchased a WioLoRaWANFieldTester field tester with battery from seeed.
After reinstalling firmware and setting GPS to 9600bds I could register on my TTN Gateway  and GPS worked.

I followed the instructions on the wiki, unfortunately, I don't understand how to set-up payload formatters for TTN mapper and TTN in general. I followed the instructions on the wiki without adding any formatters, unfortunately it replies "impossible to decode data". So I need to decode data ...

I found this webhook code and used it:

```
function Decoder(bytes, port) { 
  var decoded = {};
  
  if ( port != 1 ) return decoded;
  
  var lonSign = (bytes[0]>>7) & 0x01 ? -1 : 1;
  var latSign = (bytes[0]>>6) & 0x01 ? -1 : 1;
  
  var encLat = ((bytes[0] & 0x3f)<<17)+
               (bytes[1]<<9)+
               (bytes[2]<<1)+
               (bytes[3]>>7);

  var encLon = ((bytes[3] & 0x7f)<<16)+
               (bytes[4]<<8)+
               bytes[5];
  
  var hdop = bytes[8]/10;
  var sats = bytes[9];
  
  const maxHdop = 2;
  const minSats = 5;
  
  if ((hdop < maxHdop) && (sats >= minSats)) {
    // Send only acceptable quality of position to mappers
    decoded.latitude = latSign * (encLat * 108 + 53) / 10000000;
    decoded.longitude = lonSign * (encLon * 215 + 107) / 10000000;  
    decoded.altitude = ((bytes[6]<<8)+bytes[7])-1000;
    decoded.accuracy = (hdop*5+5)/10
    decoded.hdop = hdop;
    decoded.sats = sats;
  } else {
    decoded.error = "Need more GPS precision (hdop must be <"+maxHdop+
      " & sats must be >= "+minSats+") current hdop: "+hdop+" & sats:"+sats;
  }

  return decoded;
}
```

Also I wonder whether it is possible to use an experimental project in TTN Mapper for TTN. Can I integrate directly wih TTN Mapper and not disk91.

Kind regards,
Ffries

ffries, 2023-04-10:
Dear all,

After the wio field tester went off battery it rebooted and now it is connecting to my Helium Gateway. I don't want to connect to Helium gateways, for several reasons, including it ***might*** be a Ponsi Scheme.

When I click on survey it goes directly to Helium:
https://dev.disk91.com/ft/r/?x=48060962DB

How to remove Helium completely and revert to stock TTN network and TTN mappers? Shall I recompile the firmware. Can you provide an alternative firmware without Helium Bonzo-Bonzo.

Kind regards,
ffries

disk91, 2023-04-10:
The project, when connected to the backend dev.disk91.com automatically  pushes the data to ttnmapper.
For the rest, read the setup guide for TTN

ffries, 2023-04-12:
Thanks, closing the issue.