disk91/WioLoRaWANFieldTester #73: Are credentials lost when going off battery?

ffries, 2023-04-10:
Dear Disk91,

But what happens if battery goes off completely to 0%? All all TTN credentials lost and reverted to stock firmware?

Kind regards,
Ffries

disk91, 2023-04-10:
No they are not, credentials are stored in NVM and LoRa-E5 EEPROM

ffries, 2023-04-12:
Thanks, closing the issue.

ffries, 2023-04-12:
I understand what is going on : credentials are locked. I tried to remove the lock and use the lastest firmware. It reboots but credentials are still locked.

There should be a way to store credentials from the UI or using AT commands.

ffries, 2023-04-12:
Dear Disk91,

Thank you for your answer. I am testing TTN coverage and I could connect to TTN using AT commands. This worked well. However, when switching-off WioLoRaWANFieldTester credentials are back to Helium.

Kind regards,
FFries

disk91, 2024-05-29:
Hello, sorry to answer so late, I missed your response.
You can follow the guide in documentation to unlock your device and definitly remove the seeed factory configuration.

disk91, 2024-05-29:
see: https://github.com/disk91/WioLoRaWANFieldTester/issues/34

closed 2024-06-09 by disk91