disk91/WioLoRaWANFieldTester #84: chirpstack application template

disarticulate, 2024-06-29:
### got Chirpstack v4 running, but can't see to connect after setting up the  device in Arduino IDE:
Have A Wio Terminal Tester with https://wiki.seeedstudio.com/Wio_Terminal_LoRaWan_Chassis_with_Antenna-built-in_LoRa-E5_and_GNSS_EU868_US915
Cleared the Eprom with Clear_LoRaE5_eprom.uf2
Then setup as
```
16:08:24.697 -> LoRaWan configuration OKZONE: US915
16:16:25.394 -> OK
16:16:25.394 -> KO
16:16:25.394 -> KO
16:17:08.925 -> DEVEUI:2CF7F1C054100641
16:17:08.925 -> OK
16:19:44.497 -> APPKEY:564xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
16:19:44.497 -> OK
16:20:11.858 -> APPEUI:D2BF2FD0069ED850
16:20:11.858 -> OK
16:20:11.894 -> LoRaWan configuration OK
```
Default settings I got from default templates looks like:
[image]

Device config:
[image]

I also setup a sensecap T1000 and that actived, so there's something I'm missing with setup conditions.

Any help would be appreciated. The gateway shouldn't matter but its Dragino DLOS8N

disk91, 2024-06-29:
Did you restarted the device after configuration ?
What do you see in LoRaWan Frame panel ?

disarticulate, 2024-06-29:
Yes, it's been restarted.

[image]+
[image]
Frames.
[image]